asrob-uc3m/robotDevastation #20: Fix issues on http://asrob.uc3m.es/rddoc/todo.html

jgvictores, 2015-02-09:
Fix issues on http://asrob.uc3m.es/rddoc/todo.html, and drop use of @todo in favor of GitHub issues.

David-Estevez, 2016-04-06:
I'm commenting here the issues to remove the need to go to doxygen to check them out:

### Class BasicImageManager
Move this class to RdImageLib, set it to be compiled only with the tests 

### Member rd::RdGameScreen::RdGameScreen ()
Fix this (to return something in case of error) 

### Class RdMockupImageEventListener
Move this class to RdImageLib, set it to be compiled only with the tests

David-Estevez, 2016-04-06:
TODOs are now included in the repo as issues, closing this issue.

jgvictores, 2016-11-27:
Look what the cat dragged in...  http://asrob.uc3m.es/rddoc/todo.html 😺

David-Estevez, 2016-11-27:
That is under control, it was documented in #34. Therefore, I'm closing this issue again.

closed 2016-11-27 by David-Estevez