asrob-uc3m/robotDevastation #21: Check modules that are enabled with cmake

David-Estevez, 2015-02-09:
Some modules cannot be disabled in cmake, as they are dependencies to other modules.  

We should check these variables and enable only logic combinations (e.g. disabling tests, disabling main program, differenciating between tests and mockup tests, etc).

PeterBowman, 2017-03-23:
@David-Estevez these combinations (I guess you refer to [`cmake_dependent_option()`](https://cmake.org/cmake/help/v3.0/module/CMakeDependentOption.html) commands and of the like) are not implemented, I wonder if they ever were. Could you please expand on your initial thoughts? Are those concerns still valid today?

See also #95.

David-Estevez, 2017-03-23:
In some old version of Robot Devastation, we had all different modules listed as ENABLE_X options, which didn't make sense because most of those modules were related and depended on each other, and that was the reason of this issue.

The solution I had in mind was to allow disabling blocks of modules (i.e.  allowing the user to disable the compilation of the `RdServer` if you only wanted to compile the main game.

My old self didn't know about that option, so the implementation I I was thinking then was  to use simple if() cmake commands.

PeterBowman, 2017-03-23:
Thanks for the clarification, @David-Estevez. I rewinded the repo to early 2015 and couldn't find those ENABLEs, so I concluded that such modularization never took place.

We'l have to play with the currently available options:

* `ENABLE_robotDevastation`
* `ENABLE_rdServer`
* `ENABLE_tests`

I'm going to mark this issue as blocked (superseded?) by #95 since any planned/ongoing work would probably clash with the modifications introduced here.

PeterBowman, 2017-05-04:
Note to self: use [`cmake_dependent_option()`](https://cmake.org/cmake/help/v3.0/module/CMakeDependentOption.html) to handle dependencies? (i.e. don't compile tests alone, force to build together with RD libraries)

PeterBowman, 2018-08-02:
I'd like to apply https://github.com/roboticslab-uc3m/questions-and-answers/issues/54 here, but we should first decide on the approach for modularization and mandatory/optional components, i.e. it wouldn't make sense to require SDL if I just want to compile the server app.

PeterBowman, 2018-09-04:
> I'd like to apply roboticslab-uc3m/questions-and-answers#54 here, but we should first decide on the approach for modularization and mandatory/optional components, i.e. it wouldn't make sense to require SDL if I just want to compile the server app.

Done at https://github.com/asrob-uc3m/robotDevastation/commit/45d01faad4124c910fe8edff2ce9edd88fbd8a9b just to keep things tidy. We don't know yet how to proceed with the issue described here.