asrob-uc3m/robotDevastation #23: Encapsulate damaging player inside RdPlayer class

David-Estevez, 2015-02-10:
Currently there is no explicit way of damaging a player. It involves retrieving the current level of heath and setting the new one. 

I think this should be encapsulated.

David-Estevez, 2016-04-26:
There is also no way of restoring health of player through mentalMap. I had to do nasty things to solve this when respawning dead player:

```c++
        //-- Restore things (health, enable stuff)
        //-- Note: this is a workaround because one cannot modify player's health through
        //-- mentalMap directly (yet)
        std::vector<RdPlayer> players = mentalMap->getPlayers();
        int id = mentalMap->getMyself().getId();
        for (int i = 0; i < players.size(); i++)
            if (players[i].getId() == id)
            {
                players[i].setHealth(MAX_HEALTH);
                break;
            }
        mentalMap->updatePlayers(players);
```

David-Estevez, 2016-06-22:
To solve this, I propose the following addition to the MentalMap API:
```cpp
class MentalMap
{
    //-- Restores the health of current player (and does more stuff if needed)
    bool respawn();
}
```

And the following addition to RdPlayer API:

```cpp
class RdPlayer
{
   //-- Reduces this player's health by an amount specified by the weapon used
   bool getDamageFromWeapon(RdWeapon& weapon);
}
```

jgvictores, 2016-06-23:
Seems reasonable.

David-Estevez, 2016-06-30:
Fixed in 469edeb

closed 2016-06-30 by David-Estevez