asrob-uc3m/robotDevastation #30: SDLAudioManager not reaching RdMentalMap

jgvictores, 2016-02-10:
OS: Ubuntu trusty
Branch: develop
SDLAudioManager not reaching RdMentalMap. SDLAudioManager used in RobotDevastation class, but is null in RdMentalMap.

David-Estevez, 2016-03-23:
For some reason, when the AudioManager was requested in MentalMap's constructor, the SDLAudioManager was not registered yet. 

As a good practise from now on, it is advised to register the manager that is going to be requested in each constructor, except if several can be used (i.e. mockup, for testing purposes) and the default Manager is requested.

Fixed at 81d19f8

jgvictores, 2018-04-06:
Reopening, as 81d19f8739ac1071f90368ff1c0d93948819b94d was more of a work-around than a fix.