asrob-uc3m/robotDevastation #31: Easy stuff for dummies

Rodriwp, 2016-02-10:
This issue is pretended to have easy stuff for beginners that wants to start developing on robotDevastation. Also is posible to have an easy tag for that issue that are at an entry level ?

David-Estevez, 2016-02-10:
I suggest creating a new repo called "robotDevastation-research" (or something similar) in which we research and develop apps related to augmented reality, tracking, etc that will be later incorporated to future versions of the main robotDevastation

jgvictores, 2016-02-13:
+1 to "robotDevastation-research" !!

David-Estevez, 2016-02-15:
Suggested lines:
* Add more cool robots to RobotDevastation:
    * https://github.com/bqlabs/loki
    * https://github.com/bqlabs/mizu
* Add AR interfaces:
    * http://www.slashgear.com/reality-editor-ar-app-connects-iot-devices-by-drawing-lines-15418452/
    * http://www.realityeditor.org
    * http://www.openhybrid.org/adding-object-and-marker.html
* Research about Game Engines:        
    * Unity
    * Unreal
    * [Ogre](http://www.ogre3d.org/)
    * [Irrlicht](http://irrlicht.sourceforge.net/)
    * [Crystal Space](http://www.crystalspace3d.org/main/Main_Page)

jgvictores, 2016-02-15:
New repository created: https://github.com/asrob-uc3m/robotDevastation-research

closed 2016-02-15 by jgvictores