asrob-uc3m/robotDevastation #45: FSM integration in game

David-Estevez, 2016-05-12:
Now that the FSM for the game is implemented, it needs to be integrated in the game (substituting old code).

David-Estevez, 2016-05-17:
Behavior of RdYarpNetworkManager is not the same as MockupNetworkManager, so it has to be updated to match (so that the game can work).

David-Estevez, 2016-05-29:
SDL stuff is now fixed (#52), game seems to fail once the login in the server has been performed.

David-Estevez, 2016-05-31:
As FSM->start() is non-blocking, the program was performing cleanup just right after starting the FSM in the configure() method of robotDevastation module.

As the FSM has its own threads running, the main thread of the RFModule can be used to monitor the current state of the FSM, and to trigger cleanup when the FSM has arrived the end state.

David-Estevez, 2016-05-31:
Done in a032880f5764a304e6d1e1d3a7f43e7f1b98a74b , but requires testing with the actual robot(s) in order to check that everything works as expected.

David-Estevez, 2016-06-20:
Last time we checked, images were not being received correctly from the robot (only the first frame capture was being displayed). 

This requires more testing for validation.

jgvictores, 2016-10-19:
Tested and working.

closed 2016-10-19 by jgvictores