asrob-uc3m/robotDevastation #46: Add setup of preconditions in FSM test using previous states directly

David-Estevez, 2016-05-12:
As it is done in testGameState, all preconditions can be setup by using previous states code directly. This is error-proof, as the previous states are also covered by unit tests.

This is done in testGameState this way, and could be done in testDeadState in a similar way.

PeterBowman, 2017-10-20:
@David-Estevez could you take a look here and check whether this issue has been resolved since then? The snapshot of the unit tests directory at the time of opening this is: https://github.com/asrob-uc3m/robotDevastation/tree/8eb14b2/test.

David-Estevez, 2017-10-21:
Checking `testDeadState.cpp`, lines 139-146:
https://github.com/asrob-uc3m/robotDevastation/blob/8eb14b2b3518e34ea7b301592b90a10c933879b1/test/testDeadState.cpp#L139-L147

it seems that this has not been solved yet.