asrob-uc3m/robotDevastation #49: Add Tests to Travis

David-Estevez, 2016-05-18:
Add Tests to Travis

jgvictores, 2016-05-19:
I've opened https://github.com/asrob-uc3m/robotDevastation-developer-manual/issues/1 on documenting what each test does in the Developer Manual.
Maybe this is overkill and should be done in doxygen?

David-Estevez, 2016-05-19:
Could be interesting to know what tests are automated an which ones are manual

David-Estevez, 2016-05-29:
Tests have been added, but not all of them work on travis. Errors should be checked to update the travis config file.

jgvictores, 2016-11-23:
> Could be interesting to know what tests are automated an which ones are manual

With this we mean we should categorize tests:
- Which ones require physical robots
- Which ones require graphics
- Which ones can be run on Travis

David-Estevez, 2016-12-20:
Now only automated tests are run by travis, but we have still errors related to #77 and #86. As the scope of this issue was to set the correct tests on travis, I'm closing this issue.

Tests are now documented [in the developer manual](https://asrob-uc3m.gitbooks.io/robotdevastation-developer-manual/content/en/tests.html), only in English at this moment.

closed 2016-12-20 by David-Estevez