asrob-uc3m/robotDevastation #51: Images resources searched with a local path

David-Estevez, 2016-05-20:
Images resources searched with a local path (use ResourceFinder instead).

@jgvictores juansays: segfault if not found

David-Estevez, 2016-06-16:
@jgvictores which is the default folder in which the resources are installed / looked for?

For instance, which folder corresponds to the following piece of code? 

```cpp
yarp::os::ResourceFinder rf;
rf.findFileByName("../sounds/RobotDevastationBSO.mp3")
```

jgvictores, 2016-06-16:
`yarp::os::ResourceFinder` is a rather complex, as seen in https://github.com/robotology/yarp/issues/336, which points to "the painful details http://wiki.icub.org/wiki/YARP_ResourceFinder"

Apart from that piece of code, two intermediate important of lines from [robotDevastation/main.cpp](https://github.com/asrob-uc3m/robotDevastation/blob/d537ac4a8ac3b86c6659feed3633439822d16ea7/src/programs/robotDevastation/main.cpp#L73-L74) affect ResourceFinder. Combined, this results in:
```cpp
yarp::os::ResourceFinder rf;
rf.setDefaultContext("robotDevastation");
rf.setDefaultConfigFile("robotDevastation.ini");
rf.findFileByName("../sounds/RobotDevastationBSO.mp3");
```
The ```yarp_install``` CMake directive copies/installs the files so they can be found by the ResourceFinder. Following the RobotDevastationBSO.mp3 example, the relevant lines of [robotDevastation/share/sound/CMakeLists.txt](https://github.com/asrob-uc3m/robotDevastation/blob/d537ac4a8ac3b86c6659feed3633439822d16ea7/share/sounds/CMakeLists.txt) are:
```cmake
set(appname sounds)
file(GLOB mp3s ${CMAKE_CURRENT_SOURCE_DIR}/*.mp3)
yarp_install(FILES ${mp3s} DESTINATION ${RD_CONTEXTS_INSTALL_DIR}/${appname})
```
On my configuration, ```yarp_install``` copies/installs the files so they can be found by the ResourceFinder **relative** to ```robotDevastation/build/share/rd/contexts/robotDevastation``` or ```/usr/local/share/rd/contexts/robotDevastation/``` (note that this is where the ```robotDevastation.ini``` file the third line of cpp above specifies lies).

David-Estevez, 2016-06-16:
@jgvictores Fonts are currently accessed though their absolute path (i.e: 
```cpp
const std::string rd::DeadScreen::FONT_PATH = "/usr/share/fonts/truetype/freefont/FreeMono.ttf";
```
)

Is this ok or does it involve any risks?

jgvictores, 2016-06-16:
```cpp
const std::string rd::DeadScreen::FONT_PATH = "/usr/share/fonts/truetype/freefont/FreeMono.ttf";
```
Looks totally dangerous and non-portable. If size and license are not an issue, I'd at least think of distributing it with robotDevastation.

David-Estevez, 2016-06-16:
Everything fixed in 00f62ce

David-Estevez, 2016-06-16:
Path should be something like "../images/image.png"

David-Estevez, 2016-06-20:
Fixed in fa4dd65

closed 2016-06-20 by David-Estevez