asrob-uc3m/robotDevastation #53: Virtual Machine RobotDevastation

Rodriwp, 2016-05-26:
In order to make easy starting with robot devastation, it will be nice to create a virtual machine with robotdevastation and developer tools.

Probably the contents of the machine should be:

- Qt creator 
- RobotDevastation
- Git 

I subject to use Ubuntu Gnome

David-Estevez, 2016-05-26:
In order to use Robot Devastation easily, the proper solution would be to distribute debian packages (and windows installer). But that only makes sense with a proper playable version, which we currently don't have because of issues.

For developers (and from my point of view) installing the few things required to work on Robot Devastation is enough and very easy if they follow the existing instructions. If someone is having trouble with just the dev tools, he/she will definitely struggle with the code, so there is no point in lowering the entry barriers in that case.

jgvictores, 2016-05-28:
What we actually have to work on is: https://github.com/asrob-uc3m/robotDevastation-developer-manual

jgvictores, 2016-05-29:
BTW, CPack could be used for distribution, e.g. CPackDeb for .deb generation.

David-Estevez, 2016-05-30:
I really like this image: 

[image] 

From the [Jupyter installation instructions](http://jupyter.readthedocs.io/en/latest/projects/content-projects.html#configuration). I think it would be great to have something similar in the main page of our wiki.

closed 2022-06-04 by PeterBowman