asrob-uc3m/robotDevastation #56: XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":0"

jgvictores, 2016-06-22:
This bug seems more recent than #1. Happens mostly after initialization, after first image has arrived.
```
XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":0"
      after 714 requests (714 known processed) with 0 events remaining.
Assertion 's' failed at pulse/stream.c:1662, function pa_stream_writable_size(). Aborting.
Aborted (core dumped)
```
Pretty consistent on Ubuntu machines.

David-Estevez, 2016-09-28:
If I use a MockupInputManager instead of RdSDLInputManager, the program won't crash. So I deduce there is some kind of problem with the SDL manager.

Another intriguing fact is that if I open the game and don't press any key, the program won't crash, but the moment I press any key to log in, even if I don't press another key, the program ends up crashing.

jgvictores, 2016-09-28:
On gdb haven't seen the pulse/stream.c line. Anyway, here's the backtrace:
```
(gdb) bt
#0  0x00007ffff6593b31 in XFlush () from /usr/lib/x86_64-linux-gnu/libX11.so.6
#1  0x00007ffff7b7b470 in ?? () from /usr/lib/x86_64-linux-gnu/libSDL-1.2.so.0
#2  0x00007ffff7b7cb0b in ?? () from /usr/lib/x86_64-linux-gnu/libSDL-1.2.so.0
#3  0x00007ffff7b53e6c in SDL_PumpEvents () from /usr/lib/x86_64-linux-gnu/libSDL-1.2.so.0
#4  0x00007ffff7b53ea9 in SDL_PollEvent () from /usr/lib/x86_64-linux-gnu/libSDL-1.2.so.0
#5  0x000000000047cbbd in rd::RdSDLInputManager::update (this=0x9d5900) at /home/yo/repos/robotDevastation/src/libraries/RdInputLib/RdSDLInputManager.cpp:64
#6  0x000000000047cdb2 in rd::RdSDLInputManager::run (this=0x9d5900) at /home/yo/repos/robotDevastation/src/libraries/RdInputLib/RdSDLInputManager.cpp:110
#7  0x00007ffff742f57d in RateThreadCallbackAdapter::run() () from /usr/local/lib/libYARP_OS.so.1
#8  0x00007ffff74753f5 in theExecutiveBranch(void*) () from /usr/local/lib/libYARP_OS.so.1
#9  0x00007ffff367ab16 in ACE_OS_Thread_Adapter::invoke() () from /usr/lib/libACE-6.0.3.so
#10 0x00007ffff4720184 in start_thread (arg=0x7fffe9815700) at pthread_create.c:312
#11 0x00007ffff5b5637d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

David-Estevez, 2016-10-30:
I am going to assume issue #71 solved this at  27b941dd116ba1e

If this happens again, we will reopen the issue

closed 2016-10-30 by David-Estevez

xricson, 2017-07-27:
"" (XServer crash) 

Наконец я посмотрел на ваше бинго. Все исходит из того, 
что интерфейс ядра для nvidia-current не может быть создан 
из-за отсутствия заголовков. «Apt-get purge» все, что вы установили, 
применительно к nvidia-current, затем начните это с:

sudo apt-get install linux-source 
sudo apt-get install linux-headers-generic 
sudo apt-get install linux-image 
sudo apt-get install nvidia-current 
jockey-text -e xorg:nvidia_current
sudo reboot

Основной материал на этой странице:
https://wiki.ubuntu.com/DebuggingProgramCrash