asrob-uc3m/robotDevastation #60: Strangely large amount of Key Release while Key is pressed

jgvictores, 2016-06-29:
During robotDevastation, there is a strangely large amount of Key Release detections **while** a Key is pressed. Consequences:
- Robot receives too many stop() commands.
- May be a cause of #56.

Note: This does not happen in the testRdInputManager.

jgvictores, 2016-10-13:
All these issues seem related. Our current theories are related to the fact that [SDL_PumpEvents](https://wiki.libsdl.org/SDL_PumpEvents) is being called by threads that are not the main video thread (see [SDL_PollEvent](https://wiki.libsdl.org/SDL_PollEvent) in `RdInputLib/RdSDLInputManager.cpp`.

Must experiment with events such as [SDL_AddEventWatch](https://wiki.libsdl.org/SDL_AddEventWatch) so a callback can call the listeners.

David-Estevez, 2016-10-30:
I am going to assume issue #71 solved this at  27b941dd116ba1e

If this happens again, we will reopen the issue

closed 2016-10-30 by David-Estevez