asrob-uc3m/robotDevastation #61: Virtual robot for testing software

David-Estevez, 2016-09-25:
While developing software, usually a robot is required to launch the Robot Devastation game for testing. But most of the time we are developing at home, or we have robots, but the batteries are dead.

I propose developing a simple virtual robot that works with a USB camera attached to the players PC and can be launched anywhere with no need for the actual robot, and the program should recognize it as a real robot.

jgvictores, 2016-09-26:
I wouldn't call it a virtual robot... 🤔

David-Estevez, 2016-09-26:
I'm open to a new name for this kind of robot.

David-Estevez, 2016-09-26:
What I meant was a robot, similar to the example described in: http://asrob.uc3m.es/index.php/C%C3%B3mo_empezar_con_Robot_Devastation#Sin_robot but with images coming from the webcam.

jgvictores, 2016-09-26:
Okay, I've changed the issue title to adjust to that.

David-Estevez, 2016-09-26:
will you take care of this one?

jgvictores, 2016-09-26:
I've self-assigned, but may need help as it involves mockupImageManager which I believe has changed since my original commits.

jgvictores, 2016-09-28:
Finished at 6a6cfa96dd02a3bd06ccf8680455fab5ba1b56dd
Documented at http://asrob.uc3m.es/index.php/C%C3%B3mo_empezar_con_Robot_Devastation#Sin_robot_y_con_webcam_del_PC

jgvictores, 2016-10-15:
This issue should remain open until PC local installation of used YARP `opencv_grabber` plugin is documented.

jgvictores, 2016-10-15:
Finished at 8b8218fac76856456f6be40363c795a3f90c0b63

closed 2016-10-15 by jgvictores