asrob-uc3m/robotDevastation #70: Move SDL_Quit from atexit to cleanupSDL ?

jgvictores, 2016-10-16:
We usually have an empty `cleanupSDL`, whereas `SDL_Quit` is called from `atexit` (which I assume we control less, potentially causing segfaults at exit).

Would it be reasonable to move `SDL_Quit` from `atexit` to `cleanupSDL`?

David-Estevez, 2016-10-16:
What would then happen when closing the SDL window? ```SDL_Quit``` might be required to be called also from the main thread (which I guess is enforced by using ```atexit```)

jgvictores, 2016-10-17:
Even if closing an SDL window weren't caught as a `signal` by the RFModule, we could catch the https://wiki.libsdl.org/SDL_WindowEvent and start closing the program cleanly.

PeterBowman, 2016-11-24:
@David-Estevez 
> What would then happen when closing the SDL window? SDL_Quit might be required to be called also from the main thread (which I guess is enforced by using atexit)

For some reason I'm unable to close the window, not even by right-clicking on the Unity tile and selecting *Quit*. Does it work for you?

jgvictores, 2016-11-25:
Is there any reason we are not getting rid of the atexit() on the tests?
The `FIXME` comment is even more cryptic to me...

David-Estevez, 2016-11-25:
@PeterBowman I cannot close the window either, I can only finish the program by pressing the ESC key. 

It might just be that SDL2 requires us to bind the event to some function, but could you please open a new issue with this bug?

PeterBowman, 2016-11-25:
@jgvictores since we are not encapsulating SDL resouce handlers in an object, it made sense to me to avoid calling `SDL_Quit` on *every* exit condition (if you look at [testSDLTextDisplay.cpp](https://github.com/asrob-uc3m/robotDevastation/blob/082f69f/test/testSDLTextDisplay.cpp#L14), there are five of them given a successful SDL initialization). I treated this as a temporal hack under the assumption that we eventually want to get rid of `atexit`, hence those `FIXME`s pointing to this issue for future considerations.

@David-Estevez see #81.

jgvictores, 2016-11-27:
Maybe we could add some `int cleanupSDL()` function in the tests to be called at each exit condition, that additionally groups whatever we decide on #80.

jgvictores, 2016-11-27:
1. May main consideration with `atexit` was a possible conflict with the fact that robotDevastation already has exit handlers. However, in the tests, while I do think it obfuscates code a bit, it does not look potentially evil.
2. Regarding `FIXME` among code, it's something we try to avoid (I've just touched [this](https://github.com/roboticslab-uc3m/best-practices/commit/dfdc1277bf316ff5824045628e7b8214ea5f8c08), and also related: #20). So, we have the option to remove the `FIXME` tag and close the issue, or consider `atexit` an unwanted practice and leave this issue open until we change it. I'm okay with either option! Your move!

David-Estevez, 2016-11-27:
It is very likely that those tests with several exit conditions are the ones that cannot be automated with travis, so if it simplifies the code I don't mind to have some instances of `atexit` around. But if you think it is wiser to unify the cleanup method, it is also ok to add `cleanupSDL`.

With respect to point 2, if the issue the *FIXME* or *TODO* refers to in the code is correctly documented as an issue, I don't mind having that tag in the code. Especially if it helps the person in charge of the issue to locate the problem and fix it faster when working in that issue later.

jgvictores, 2016-11-30:
1. So we're good to go with `atexit`!
2. I've updated [this](https://github.com/roboticslab-uc3m/best-practices/commit/b688e2cfc538a28d419bef1ab92358df1340d9d8) to say it's okay there's a `toDo` or a`fixMe` as long as there is an associated **open issue** (referenced to each other).

Given 1 and 2, in this case we can just remove the `fixMe`s and close the issue.

PR #79 (2016-11-25): Call SDL_Quit on object destruction instead of exit()
Remaining atexit() occurrences were kept until a better solution is found. Also add missing return values to tests and remove superfluous includes of &lt;cstdlib&gt;. Fixes #70.
commit cc15e064 (2016-11-30): Remove unused FIXMEs

Closes #70, we'll keep those atexits around.
closed 2016-11-30 by PeterBowman