asrob-uc3m/robotDevastation #75: Major memory leak

David-Estevez, 2016-11-09:
When running robotDevastation, RAM consumption increases up to 6GB in my laptop and 9.1GB in @jgvictores 's laptop.

This might be due to not freeing SDL surfaces after drawing them.

David-Estevez, 2016-11-09:
Some clues: http://forums.libsdl.org/viewtopic.php?t=6534&sid=ca7be0700455ef5a09f10a9e683d6374

It might be the TTF renderer function that is creating surfaces without deleting the old ones. An example from that forum:

```
string text;
SDL_Surface *text = TTF_RenderText("Initial String");

while(running)
{
    if(textUpdated())
    {
      SDL_Surface *temp = TTF_RenderText(text);
      if(temp)
      {
          SDL_FreeSurface(text);
          text = temp;
      }
      else { /* handle error */ }
    }
    SDL_BlitSurface(text, screen);
    SDL_Flip(screen);
}

SDL_FreeSurface(text);
```

David-Estevez, 2016-11-15:
I added several `SDL_FreeSurface` statements (at 4dcd424cbccab8696550bc58c77d89baf87b7efa) but the memory leak seems to persist.

I have noticed that the memory consumption increases in steps of ~6-7 MB, which is consistent with the image size (640*480*3*8 bits ~= 7MB) either from the SDL screen (unlikely since I'm freeing all surfaces in use) or from the yarp image / zbar.

jgvictores, 2016-11-19:
Please do not BLAME... but...
https://github.com/asrob-uc3m/robotDevastation/blob/e4eef5b5b6ce81a95829c914759543971e743509/src/libraries/RdImageLib/RdProcessorImageEventListener.cpp#L29

David-Estevez, 2016-11-20:
Done at e42f99a

commit e42f99ac (2016-11-20): Fix #75 (malloc witthout free)
closed 2016-11-20 by David-Estevez