asrob-uc3m/robotDevastation #77: .travis.yml not updated to SDL2

jgvictores, 2016-11-14:
.travis.yml not updated to SDL2, must update.

jgvictores, 2016-11-14:
Not as easy as https://github.com/asrob-uc3m/robotDevastation/commit/7dfac6c24150486b5f6f213827ea5f2de4cb60cc (see https://travis-ci.org/asrob-uc3m/robotDevastation/jobs/175587238 )

PeterBowman, 2016-12-19:
I see two alternatives here:

* compile SDL2 from sources
* upgrade Travis host to trusty (beta): https://docs.travis-ci.com/user/trusty-ci-environment/

PeterBowman, 2016-12-19:
Some progress was made with the second approach and [jobs/185271003](https://travis-ci.org/asrob-uc3m/robotDevastation/jobs/185271003).

David-Estevez, 2016-12-19:
I think the approach you are working in is the most sensible one.

PeterBowman, 2016-12-24:
Done in 8385e8d.

@jgvictores your commit has been cherry-picked in 48fa41c and merged into develop, you might want to delete the `issue-77-travis-sdl2` branch now.

closed 2016-12-24 by PeterBowman

jgvictores, 2016-12-24:
@PeterBowman Yes, deleted the `issue-77-travis-sdl2` branch! Thanks as always!