asrob-uc3m/robotDevastation #78: This side up for QR codes

jgvictores, 2016-11-23:
Is this a bug or a feature? :smile:

David-Estevez, 2016-11-23:
I would say that right now it is OK, but it could be improved by supporting rotated QR codes.

jgvictores, 2016-11-24:
It was just a fun way to document the fact that we currently do not support rotated QR codes. 🙃

PeterBowman, 2020-07-07:
@jgvictores I know we considered refactoring QR code detection into a new module/device at some point, but I'm not sure where those discussion happened (asrob/rl?). Anyway, note that OpenCV 4.0 introduces QR code handling ([ref](https://github.com/opencv/opencv/wiki/ChangeLog#version400)):

> QR code detector and decoder have been added to opencv/objdetect module together with live sample. Currently, the decoder is built on top of QUirc library (which snapshot we put into opencv). Thanks to our intern Alexander Nesterov for this important contribution!

jgvictores, 2020-07-08:
@PeterBowman thanks for the info!

Yes, my idea was to merge it into the output of https://github.com/roboticslab-uc3m/vision/issues/79 which is targeted by https://github.com/roboticslab-uc3m/vision/pull/98