asrob-uc3m/robotDevastation #80: Close *all* resources before exit

PeterBowman, 2016-11-24:
As a follow-up to #79, I'd suggest closing TTF- and IMG-related resources, too. Similarly to `SDL_Init()` and `SDL_Quit()`, the following functions apply here:

* `TTF_Quit()`: https://www.libsdl.org/projects/SDL_ttf/docs/SDL_ttf_10.html#SEC10
* `IMG_Quit()`: https://www.libsdl.org/projects/SDL_image/docs/SDL_image_9.html#SEC9

If I understand it correctly, both should be called immediately before `SDL_Quit()`. We should have now a proper place to call these functions from: `rd::SDLScreenManager::cleanupSDL`.

PeterBowman, 2016-12-01:
Started with 2da58f1. Pasting [`atexit(3)`](http://man7.org/linux/man-pages/man3/atexit.3.html) function description for future reference (bolding mine):

> The atexit() function registers the given function to be called at normal process termination, either via exit(3) or via return from the program's main().  **Functions so registered are called in the reverse order of their registration**; no arguments are passed.

PeterBowman, 2016-12-05:
Merged. Note that it's safe to call *_Quit functions even if no modules have been initialized yet.

closed 2016-12-05 by PeterBowman

jgvictores, 2016-12-05:
@PeterBowman Thanks! Great work!!

David-Estevez, 2016-12-06:
Cool! Thanks, @PeterBowman !!