asrob-uc3m/robotDevastation #81: SDL window is unresponsive to usual closing events other than pressing "Esc"

PeterBowman, 2016-11-25:
https://github.com/asrob-uc3m/robotDevastation/issues/70#issuecomment-262845721
> For some reason I'm unable to close the window, not even by right-clicking on the Unity tile and selecting Quit. Does it work for you?

https://github.com/asrob-uc3m/robotDevastation/issues/70#issuecomment-262985229
> I cannot close the window either, I can only finish the program by pressing the ESC key. It might just be that SDL2 requires us to bind the event to some function (...)

PeterBowman, 2016-11-27:
An initial approach was taken with 5c97fb2 by adding a clause for window events at `rd::RdSDLInputManager::inputCallback`.

Temporarily applying `question` label for feedback on a new interface method in `rd::RdInputEventListener`: [6bd2ab2](https://github.com/asrob-uc3m/robotDevastation/commit/6bd2ab23f56ec9f6a0665e4f96c5e3952add5d43#diff-16da6dc17f7751df5696f00e497b945d). The window should be responsive now for both targeted actions: close button (title bar) and *Quit* on Unity.

jgvictores, 2016-11-27:
Rather than:
`virtual bool onWindowEvent( SDL_WindowEventID id ) = 0;`
Could we pass something more like the following?
`virtual bool onWindowEvent( const SDL_WindowEvent & windowEvent ) = 0;`

Not completely sure about this one, completely open to options. It just seems that the listeners may use the data that may not be accessible if we only pass the ID.

David-Estevez, 2016-11-27:
One of my main concerns is that with these changes the encapsulation is broken in RdEventListener.

Could it be possible to create a simple class RdWindowEvent such as this?

```cpp
class RdWindowEvent 
{
    public:
        virtual int getEvent() = 0;
        static const int WINDOW_CLOSED;
}
```

Then, the SDLWindowEvent could be created inheriting from that class:

```cpp
class SDLWindowEvent : public RdWindowEvent
{
    public:
        SDLWindowEvent(const SDL_WindowEvent & windowEvent);
        virtual int getEvent();
}
```

And the encapsulation of the RdInputEventListener could be maintained:
```cpp
virtual bool onWindowEvent( RdWindowEvent windowEvent ) = 0;
```
What do you think about this?

PeterBowman, 2016-12-01:
@David-Estevez this is currently in progress at ea5973e.

BTW thanks to d92221e, we could now get rid of all those pesky pointers & dynamic memory management in [rd::RdSDLInputManager::inputCallback](https://github.com/asrob-uc3m/robotDevastation/blob/d92221e/src/libraries/RdInputLib/RdSDLInputManager.cpp#L72), that is:

```c++
RdSDLKey key(event->key.keysym.sym);

if ( !(key.isPrintable() || key.isControlKey()) )
    return false;

for ( int i = 0; i < (int)listeners.size(); i++)
    listeners.at(i)->onKeyUp(key);
```

instead of

```c++
RdKey * key = new RdSDLKey(event->key.keysym.sym);

if ( !(key->isPrintable() || key->isControlKey()) )
{
    delete key;
    return false;
}

for ( int i = 0; i < (int)listeners.size(); i++)
    listeners.at(i)->onKeyUp(*key); 

delete key;
```

Are you OK with this change?

David-Estevez, 2016-12-01:
Perfect! Proceed with the changes! :+1:

PeterBowman, 2016-12-04:
I'm quite confident that `rd::RdKey` was supposed to be an abstract base class, which is not the case ATM. @David-Estevez could you confirm this?

David-Estevez, 2016-12-04:
It is abstract in the sense that you are not allowed to create instances of that class, but it provides a basic implementation of getters and operators that should behave similarly in all subclasses.

Do you think that class should have virtual members instead? Otherwise it is ok to me as it is, but I'd like to know your opinion about the implementation.

PeterBowman, 2016-12-04:
I wouldn't call it an abstract class per se. We are indeed restrained by the protected constructor and cannot invoke it outside the inheritance chain of that class (`RdKey` itself and its subclasses); however, there are several places in the code where actual instantiation occurs by means of the [implicitly-defined copy constructor](http://en.cppreference.com/w/cpp/language/copy_constructor) and [object slicing](https://en.wikipedia.org/wiki/Object_slicing), e.g.:

* [`RdSDLInputManager::inputCallback`](https://github.com/asrob-uc3m/robotDevastation/blob/cc15e06/src/libraries/RdInputLib/RdSDLInputManager.cpp#L88) + [`GameState::onKeyDown`](https://github.com/asrob-uc3m/robotDevastation/blob/cc15e06/src/libraries/GameStatesLib/GameState.cpp#L158): `RdSDLKey` is passed  **by value** and the function accepts a `RdKey` (base class), therefore an instance of the latter is created.
* [`GameState::KEY_EXIT` et al.](https://github.com/asrob-uc3m/robotDevastation/blob/cc15e06/src/libraries/GameStatesLib/GameState.cpp#L6): these are all instances of `RdKey` despite passing through `MockupKey`'s constructor to initialize member variables.

It's immediately clear that polymorphism rules won't have any effect on these cases as objects created this way will lose all information regarding the derived types (i.e. all subclass-specific member fields and overriden methods). It boggles me that neither `RdSDLKey` nor `MockupKey` define that kind of stuff, leading to the suspicion that both have been designed with object slicing in mind. If that was the case, it should be documented somewhere (it's acknowledged as an issue in the Wikipedia article, in fact it doesn't seem a programming technique to normally make use of). If not and slicing was not intended, having defined `RdKey` as an abstract class would even prevent such code from compiling.

My question is: should I preserve current behavior and make `RdWindowEvent` concrete (see https://github.com/asrob-uc3m/robotDevastation/issues/81#issuecomment-263121298), or go the other way around and keep all event-input base classes (`RdKey`, `RdWindowEvent`) abstract?

Side note: object slicing situations as described before can be easily avoided by using references (the first case should be solved now with d92221e). However, in some cases it's not so simple - I'm currently stuck at [MockupInputEventListener](https://github.com/asrob-uc3m/robotDevastation/blob/cc15e06/src/libraries/RdInputLib/MockupInputEventListener.hpp) as `vector`s must store pointers of the base type in order to take advantage of polymorphism.

David-Estevez, 2016-12-07:
As discussed, I think the best choice would be to discard the `MockupKey` and `RdSLDKey` classes and to make the following changes to `RdKey`:

* Refactor the constructor of `MockupKey` to be the constructor of `RdKey`. This way we can create instances of `RdKey` directly from the base class and we eliminate the need of a `MockupKey`
* Refactor `RdSDLKey` to become a [factory method](https://en.wikipedia.org/wiki/Factory_method_pattern) that takes a `SLDKeycode` and returns a `RdKey`. The only question to solve here is where to store the map with the correspondances between `SDLKeycode`  and characters or non-printable key codes, i.e. to encapsulate it within a class, make it a static variable within the factory method  or another solution that you might propose.

PeterBowman, 2016-12-07:
Ready as a draft at:
* bdffacd
* 01cf7bd

David-Estevez, 2016-12-07:
Looks ok to me!

PeterBowman, 2016-12-08:
As it was correctly pointed out by @David-Estevez (and I misunderstood at a first glance), the removal of `rd::RdKey`'s private default constructor ([diff](https://github.com/asrob-uc3m/robotDevastation/commit/bdffacd#diff-95c7f050b459974fe1cd29c73f8260f3)) could lead to accidental creation of derived classes. The key word here is "accidental" since it won't prevent such deliberate actions as calling one of the public constructors, anyway (e.g. `RdDerivedKey::RdDerivedKey() : RdKey('a') {}`). It's not possible to seal a class (=disallow subtyping) without resorting to [obscure solutions](https://www.experts-exchange.com/articles/227/Sealing-a-C-Class.html) for C++03-compliant code. On the other hand, C++11 provides an elegant way: the [`final`](http://en.cppreference.com/w/cpp/language/final) specifier.

I'd take the easier path and rely on proper documentation of current (and intended) behavior. However, opinions are welcome.

PeterBowman, 2016-12-08:
By the way, thanks to its new [converting constructors](http://en.cppreference.com/w/cpp/language/converting_constructor), we could rewrite [this](https://github.com/asrob-uc3m/robotDevastation/blob/bdffacd/src/libraries/GameStatesLib/GameState.cpp#L6):

```c++
const rd::RdKey rd::GameState::KEY_EXIT = rd::RdKey(rd::RdKey::KEY_ESCAPE);
const rd::RdKey rd::GameState::KEY_RELOAD = rd::RdKey('r');
```

into this:

```c++
const rd::RdKey rd::GameState::KEY_EXIT = rd::RdKey::KEY_ESCAPE;
const rd::RdKey rd::GameState::KEY_RELOAD = 'r';
```

and so on. Do you like explicitness?

jgvictores, 2016-12-08:
I understand there's only one argument, so either variant is quite supported in old compilers (pre-C++11). Therefore, I have no personal preference.

jgvictores, 2016-12-08:
PS: Apart from wanting to support old linux machines, I've never abandoned the idea of #12 (create Windows port), and I'd like to avoid as many as ventures possible.

David-Estevez, 2016-12-08:
I do not really see the need of further sealing the `RdKey` class, with a private default constructor it should suffice. But the `final` statement sounds like an elegant solution if you still want to seal it. 

About converting constructors, I did not know them (I'm a bit rusty on my C++11, I should update it), and I find them less verbose than the previous version of the code which, for me, is a plus. So I vote for using them.

PR #84 (2016-12-10): Review of event handlers, new window event listeners and RdWindowEvent class
Resolves #81. Includes:

* New `rd::RdWindowEvent` class. For now, it only supports `WINDOW_CLOSE` events.
* New `rd::RdInputEventListener::onWindowEvent` method, its implementation in several listener subclasses and invocation in `rd::RdSDLInputManager::inputCallback`.
* New `rd::MockupInputManager::sendWindowEvent` method.
* Deletion of `rd::MockupKey`; its constructors have been pulled up to `rd::RdKey`, which is now intended to be a sealed concrete class. Same applies to `rd::RdWindowEvent`.
* `rd::RdSDLKey` has b [... cut]
closed 2016-12-10 by David-Estevez