asrob-uc3m/robotDevastation #86: Travis cannot build YARP

David-Estevez, 2016-12-20:
As @PeterBowman discovered, YARP is not being built in Travis due to missing Eigen.

As far as I know, @PeterBowman is currently working to solve this.

jgvictores, 2016-12-20:
We could also just remove `-DCREATE_LIB_MATH:BOOL=ON` from [where it is](https://github.com/asrob-uc3m/robotDevastation/blob/4668ca3d8f13975ca8a51e2bad6d8477b95a964c/.travis.yml#L43) as it's actually not even a requirement anymore.

David-Estevez, 2016-12-20:
That is something @PeterBowman suggested yesterday, but neither of us knew for sure if LIB_MATH was still a requirement.

jgvictores, 2016-12-20:
I'm pretty sure we can remove:
 `-DCREATE_GUIS:BOOL=ON`
`-DCREATE_LIB_MATH:BOOL=ON`
`-DENABLE_yarpcar_bayer_carrier:BOOL=ON`

Regarding carriers, we could activate something like:
`-DENABLE_yarpcar_mjpeg_carrier:BOOL=ON`
or simply forget about them by removing:
`-DCREATE_OPTIONAL_CARRIERS:BOOL=ON`

PeterBowman, 2016-12-20:
You have to set -DCREATE_LIB_MATH:BOOL=OFF explicitly as it's enabled by default. I have to investigate it further, but I recall being unable to launch RD with this configuration.

PeterBowman, 2016-12-24:
Done in f7c969e and merged into develop with 8385e8d.

See also #87.

closed 2016-12-24 by PeterBowman