asrob-uc3m/robotDevastation #87: Unbreak tests on Travis

PeterBowman, 2016-12-24:
Following #49 and recently resolved compilation issues, tests are currently breaking because of several reasons:

- [ ] Errors during SDL initialization: `SDL Error: Failed to connect to the Mir Server` ([example](https://travis-ci.org/asrob-uc3m/robotDevastation/jobs/186536685#L3060))
- [ ] Multiple segmentation faults:
  - [ ] #69 
  - [ ] more?

PeterBowman, 2016-12-24:
Regarding SDL errors, some tests do still require graphics and/or access to audio files, which may not be supported by the testing environment without additional configuration. Travis stumbles right on the first script at [jobs/186550568#L3574](https://travis-ci.org/asrob-uc3m/robotDevastation/jobs/186550568#L3574):

```
$ bin/testDeadState
...
[error] SDLScreenManager.cpp:139 initSDL(): SDL could not initialize!
 SDL Error: Failed to connect to the Mir Server
...
ALSA lib confmisc.c:768:(parse_card) cannot find card '0'
ALSA lib conf.c:4248:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4248:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1251:(snd_func_refer) error evaluating name
ALSA lib conf.c:4248:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4727:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM default
[error] SDLAudioManager.cpp:14 SDLAudioManager(): AudioMixer could not be opened!
```

`testDeadState` opens a graphic window to perform its checks. Was it really meant to be run on Travis?

See also:

* http://stack1.ocim.top/38153660/running-opengl-sdl2-on-aws-ec2.html
* https://docs.travis-ci.com/user/gui-and-headless-browsers/
* https://bugs.launchpad.net/ubuntu/+source/mir/+bug/1516419

David-Estevez, 2017-01-09:
There are several tests, such as `testDeadState` that are automated with gtest, but they open a graphic window.

I can think of two different solutions:
1. Configure travis to run graphical apps
2. Create a mock object to avoid opening windows in tests.

What do you both think it would be the best solution? (cc @jgvictores )

jgvictores, 2017-01-09:
1. Configure travis to run graphical apps

...seems like the best option to me. :smile:

PeterBowman, 2017-01-09:
I agree, although it's not clear to me how to achieve that. BTW: ALSA errors should be gone now with 163a784.

PeterBowman, 2017-02-22:
SDL errors are gone with c0b8b32. Now it's time to check all tests one by one: https://travis-ci.org/asrob-uc3m/robotDevastation/jobs/204284401.

PeterBowman, 2017-02-22:
I had to put a sleep function at a1021b5 because of race conditions in https://travis-ci.org/asrob-uc3m/robotDevastation/jobs/204284401#L3631. @jgvictores suggests implementing `rd::MockupState` as a RPC responder (`yarp::os::PortReader` class).

cc @David-Estevez

PeterBowman, 2017-02-22:
The failed Travis build linked earlier corresponds to a GCC job, whereas Clang showed positive results. *After* having put the sleep function, GCC runs nice, but now Clang is stuck in an infinite loop: https://travis-ci.org/asrob-uc3m/robotDevastation/jobs/204323228#L3968. Notice two failed assertions in previous lines, which may be the cause. We might want to find a more reliable solution than the current one.

David-Estevez, 2017-02-23:
That test was always very... "brittle". Any possible improvement is welcomed. I have just opened #92 to discuss possible solutions.

PR #98 (2017-04-09): Unbreak testYarpNetworkManager
Closes #87. This patch set involves modifying the interface of two classes (`rd::YarpNetworkManager` and `rd::Server`) as well as implementing a new option in the `rdServer` module (#82). **Please review.**
closed 2017-04-09 by David-Estevez