asrob-uc3m/robotDevastation #88: Developer manual v1.0

David-Estevez, 2017-01-18:
*
[Introduction](README.md)
* What is Robot Devastation?
* Overview
* [Software](software.md)
* [Robot Devastation Overview](general-architecture.md)
* Game FSM
* [Finite State Machine](finite-state-machine.md)
* Sound Management
* Input Management
* Network Management
* [Network](network.md)
* Robot Management
* User Interface Management
* Camera Management
* Data Management
* [Tests](tests.md)
* Server
* Network Protocol
* [Robots](robots.md)
* [Rd Series](rd-series.md)
* [Premium Series](premium_series.md)
* How to contribute
* Come to our meetings!
* Issues
* Projects
* [Resources](resources.md)
* [Research](research.md)

jgvictores, 2017-01-18:
Maybe this should go... https://github.com/asrob-uc3m/robotDevastation-developer-manual/issues ?

David-Estevez, 2017-03-26:
The Developer manual for v1.0 is finished, at last!

closed 2017-03-26 by David-Estevez