asrob-uc3m/robotDevastation #89: Avoid hardcoding remote YARP device launchers

PeterBowman, 2017-02-01:
[`rd::RdYarpRobotManager::connect`](https://github.com/asrob-uc3m/robotDevastation/blob/1669607/src/libraries/RdRobotLib/RdYarpRobotManager.cpp#L73) issues a remote command on real robots, precisely:
```bash
sudo launchRaspiYarp --device RdRobotServer --subdevice RdOnePwmMotors --name /$robotName --gpios 17 27
```
Some thoughts:
* this command fails on the `rd2new` robot - the device `RdTwoPwmMotors` with parameter `--grpios 19` must be used instead (for this particular case)
* such configuration must not be hardcoded - migrate to an .ini?

PeterBowman, 2017-02-06:
As @jgvictores mentioned at our latest workshop (and I didn't fully understand back then), this code actually belongs to the robots, where a convenient background process may launch the appropriate devices on system startup, and should be removed from this repo.

PeterBowman, 2017-02-15:
This issue is blocked by asrob-uc3m/robotDevastation-robots#8, that is, we need to provide a decent solution on the robot side before the remote initialization part here can be removed.

jgvictores, 2017-02-15:
https://github.com/asrob-uc3m/robotDevastation-robots/issues/29 before https://github.com/asrob-uc3m/robotDevastation-robots/issues/8

jgvictores, 2017-03-15:
And `sudo chmod +755 run` looks like it does the trick, but requires further testing.

jgvictores, 2017-03-22:
As of https://github.com/asrob-uc3m/robotDevastation-robots/commit/c65f8bc3ce9e65db5bd3556b99c81cf495dfc816, explained in https://github.com/asrob-uc3m/robotDevastation-robots/issues/29#issuecomment-288459618, we no longer use `yarprun`, so [this section](https://github.com/asrob-uc3m/robotDevastation/blob/166960783864534ba4f7853dff8814f46c6537e0/src/libraries/RdRobotLib/RdYarpRobotManager.cpp#L89-L134), which was already optional, can essentially be removed from our code.

jgvictores, 2017-03-22:
Done at https://github.com/asrob-uc3m/robotDevastation/commit/c3436ba03e55522e35adcb72cc258702f0d8af2e.

closed 2017-03-22 by jgvictores