asrob-uc3m/robotDevastation #96: C++ code cleanup

PeterBowman, 2017-03-23:
This issue does not address a thorough review of code flow nor any programming structures. Simply put, I'd like to gather here a list of possible fixes that one may have noticed at a first glance during code development but never had the chance to properly focus on them:

- [ ] prefixed header guards: `__DEAD_STATE_HPP__` -> `__RD_DEAD_STATE_HPP__`
- [ ] `const` keyword in getters: `getName() const { ... }`
- [ ] Doxygen documentation cleanup
- [ ] missing/superfluous #includes
- [ ] prepend `rd::` namespace to variable declarations

Please add your thoughts and expand the list if new candidate items are found.

David-Estevez, 2017-03-24:
My bad, I wrote the wrong issue number in the commit message. Reopening issue :sweat_smile:

PeterBowman, 2017-03-30:
Emacs modelines can be found in approx. 50% of C++ files in the `src` directory. YARP devs completely got rid of them about a year ago (robotology/yarp@f62d487). @jgvictores should we keep them around in RD code? Are they of any use, and if so, should we stamp them on the remaining half of source files?

jgvictores, 2017-03-30:
Yes, I'm aware they removed them... I'm used to seeing them, but would be okay with their removal.

Their only use now is dictating tab/space convention, which would be much better placed in the [robotDevastation-developer-manual](https://github.com/asrob-uc3m/robotDevastation-developer-manual) (maybe you could put an issue there to remind us this?).

commit c56b4bc9 (2017-03-24): Merge branch 'fix-96-processor-image-event-api' into develop

Fixes  #96
PR #102 (2017-04-30): C++ code cleanup
Resolves #96, see issue description for a list of implemented changes.
closed 2017-04-30 by David-Estevez