asrob-uc3m/robotDevastation #99: Determine project license

PeterBowman, 2017-04-07:
Root `LICENSE` is a BSD-3-clause. Exported CMake configs and `main.cpp` files at `src/programs/` state that this is LGPL v2.1. Several CMake list files prefer GLP v2.0.

jgvictores, 2017-04-08:
Yes, we should determine a license. For now:
- I've added some history at #100 (copyright->LGPL->BSD... :joy:).
- Our GPL seems to be only in `share`.
- We apparently do not link to any external GPL.
- I've been reading on [license switching](http://stackoverflow.com/questions/1409666/switching-license-from-gpl-to-bsd-in-later-time). Since we are few core developers, I expect we are not going to have any issues for agreeing, so transition will run smoothly.

PeterBowman, 2017-04-19:
* https://www.gnu.org/licenses/gpl-faq.html#AllCompatibility
* https://www.quora.com/What-are-the-major-differences-between-GNU-LGPL-v3-and-v2-1

jgvictores, 2017-04-19:
Ok `LGPL v2.1`?

David-Estevez, 2017-04-19:
@smorante  agrees with LPGL v2.1, what changes need to be done?

David-Estevez, 2017-04-19:
[image]

PeterBowman, 2017-05-16:
Let me drop this here regarding copyright notices on top of each source file:

http://lu.is/blog/2012/03/17/on-the-importance-of-per-file-license-information/

jgvictores, 2017-05-16:
+1 to adding copyright notices (license, author(s) or at least something ASROB) on top of each source file.

Not sure if it's stated in the article, but the fact is many times it's noobs who don't worry about licenses they're unaware about, and just copy and paste files. Adding 1-2 header lines can increase traceability of where and how our files are being used, which is definitely a plus.

PeterBowman, 2017-09-28:
I've just found this article which favours GPL over LGPL licenses:

https://www.gnu.org/licenses/why-not-lgpl.html

jgvictores, 2017-10-01:
GNU are a bit (legitimately) biased towards GPL. :smile:
LGPL has the benefit of not scaring off potential industry partners, while still promoting them to improve whatever set of core libraries we provide.

jgvictores, 2017-11-17:
Need:
1. Templates for new uniform headers (for c++, cmake). 
1. Script (possibly python) for incorporating this uniform header in files.

jgvictores, 2017-11-17:
```
* Authors: See https://github.com/asrob-uc3m/robotDevastation/blob/master/AUTHORS.md
* CopyPolicy: Released under the terms of the LGPLv2.1, see https://github.com/asrob-uc3m/robotDevastation/blob/master/LICENSE
 ```

jgvictores, 2017-11-17:
```
* Authors: See AUTHORS.md at project root.
* CopyPolicy: Released under the terms of the LGPLv2.1, see LICENSE at project root.
* URL: https://github.com/asrob-uc3m/robotDevastation
 ```

PeterBowman, 2017-12-01:
See also *How to Apply These Terms to Your New Libraries* at:

https://github.com/github/choosealicense.com/blob/7213ccf/_licenses/lgpl-2.1.txt#L494-L538

PeterBowman, 2017-12-01:
Resolved at #117.

closed 2017-12-01 by PeterBowman

PeterBowman, 2017-12-02:
Strangely enough, we actually forgot to update the LICENSE file itself. Done at 99f5d02 (and I've just forgotten to [skip ci], yay!).

jgvictores, 2017-12-02:
No problem LOL!