asrob-uc3m/robotDevastation #100: History of Robot Devastation

jgvictores, 2017-04-08:
from:	Juan G Victores <jcgvicto@ing.uc3m.es>
to:	Santiago Morante <santimorante@gmail.com>
date:	Sun, Jun 17, 2012 at 7:45 PM
subject:	Re: ecro
> Hi,
> Tengo en mente un proyecto ambicioso. ¿Te gusta el WOW y/o Counter Strike? [...]
> 
> Bienvenido a MIXED REALITY ROBOT DEVASTATION: CONQUER THE WORLD!!
> 
> ¿Cómo lo ves?

from:	Santiago Morante <santimorante@gmail.com> via ing.uc3m.es 
to:	Juan G Victores <jcgvicto@ing.uc3m.es>
date:	Mon, Jun 18, 2012 at 12:47 AM
subject:	Re: ecro

> Lo veo ambicioso, pero me apunto.
> 
> Entiendo que quieres hacer una spanish version de : http://www.youtube.com/watch?v=rIfK5JFTftc

from:	Juan G Victores <jcgvicto@ing.uc3m.es>
to:	Santiago Morante <santimorante@gmail.com>
date:	Mon, Jun 18, 2012 at 10:21 AM
subject:	Re: ecro
> Sí pero en un espacio sólo delimitado por el alcance de las comunicaciones inalámbricas, y donde puedan también entrar jugadores que no disponen de robot físico. 100% distribuido, pero te conectas a una partida en marcha para compartir puntuaciones.

from:	Santiago Morante <santimorante@gmail.com> via ing.uc3m.es 
to:	Juan G Victores <jcgvicto@ing.uc3m.es>
date:	Tue, Jun 19, 2012 at 12:37 AM
subject:	Re: ecro
> [...] Yo he ido preparando la wiki...
> 
> [image]

Commit history:
- 2012-06-20: https://sourceforge.net/p/asrob/svn/105/ (primer commit trazable de RD por @smorante)(licencia copyright)
- 2012-06-21: https://sourceforge.net/p/asrob/svn/113/ (@smorante introduce README con mención a @jgvictores)
- 2012-06-23: https://sourceforge.net/p/asrob/svn/122/ (primer commit trazable de RD por parte de @jgvictores)
- 2012-06-23: https://sourceforge.net/p/asrob/svn/123/ (different approach por @jgvictores)(introduce LGPL)
- 2012-06-23: https://sourceforge.net/p/asrob/svn/140/ (approach antiguo se mueve a superseded)
- 2014-02-24: https://github.com/asrob-uc3m/robotDevastation/commit/4b37c103a466cf30022231ae5120daa8bdfe0c91 (tras pausa de 1-2 años, primer commit a github por @jgvictores sugerido por @David-Estevez)(introduce LICENSE con BSD-3 clause).
- 2014-06-01: https://github.com/asrob-uc3m/robotDevastation/commit/0e365b9078134080ee1be3a0d88dc399728ba7e0 (primer commit trazable de RD por parte de @David-Estevez)
- 2016-11-24: https://github.com/asrob-uc3m/robotDevastation/commit/107d45f792e5c5ccc7269194c5f7c03afc5c2d2c (primer commit trazable de RD por parte de @PeterBowman)

Disclaimer:
- La búsqueda de commits ha sido manual. Pude haberme saltado commits y autores. Además, no conté con ramas huérfanas, sino con merge a `master`.
- Tb pueden haber autores sin citar de otros repositorios relacionados (https://github.com/asrob-uc3m/robotDevastation-robots, etc).

jgvictores, 2017-04-08:
Good way to celebrate #100 plus Easter holidays, guys? :smile::tada::tada::tada:

PeterBowman, 2017-04-08:
Sure! Happy 100th issue/PR! 🎉 💯 🎉

David-Estevez, 2017-04-09:
And it only took us 5 years... :stuck_out_tongue_winking_eye:

closed 2017-04-19 by jgvictores