asrob-uc3m/robotDevastation #101: Cannot use webcam, invalid image errors on Windows

PeterBowman, 2017-04-10:
Tested on different configurations (32/64-bits, *official* or [forked](https://github.com/asrob-uc3m/ZBar) ZBar binaries, latest YARP devel or v2.3.68):

```
[error] AudioManager.cpp:112 Register(): AudioManager with id "SDL" was already registered
yarp: Port /robotDevastation/0/rdServer/rpc:c active at tcp://192.168.1.50:10004/
yarp: Port /robotDevastation/0/rdServer/info:i active at tcp://192.168.1.50:10005/
yarp: Sending output from /robotDevastation/0/rdServer/rpc:c to /rdServer/rpc:s using tcp
yarp: Receiving input from /rdServer/info:o to /robotDevastation/0/rdServer/info:i using tcp
[error] YarpNetworkManager.cpp:258 keepAlive(): NetworkManager has not been started
[error] YarpNetworkManager.cpp:275 keepAlive(): Keep alive failed
[error] YarpNetworkManager.cpp:275 keepAlive(): Keep alive failed
[WARNING]ServerFrameGrabber: error configuring interfaces for parsers 
yarp: Port /rd1/img:o active at tcp://192.168.1.50:10006/
yarp: No framerate specified, polling the device
yarp: Port /robotDevastation/rd1/img:i active at tcp://192.168.1.50:10007/
[ERROR]Cannot find "mjpeg" plugin (not built in, and no .ini file found for it)
[ERROR]Check that YARP_DATA_DIRS leads to at least one directory with plugins/mjpeg.ini or share/yarp/plugins/mjpeg.ini in it
Cannot load plugin from shared library ()
(No se puede encontrar el módulo especificado.

No se puede encontrar el módulo especificado.

No se puede encontrar el módulo especificado.

No se puede encontrar el módulo especificado.

)
yarp: Failed to find library support for carrier mjpeg
yarp: Could not find carrier "mjpeg"
Failure: no way to make connection /rd1/img:o->/robotDevastation/rd1/img:i
[warning] YarpLocalImageManager.cpp:32 start(): Could not connect to robot camera via mjpeg.
yarp: Receiving input from /rd1/img:o to /robotDevastation/rd1/img:i using tcp
yarp: Sending output from /rd1/img:o to /robotDevastation/rd1/img:i using tcp
[error] MentalMap.cpp:96 getMyself(): Myself not found in mental map
[error] MentalMap.cpp:97 getMyself(): Returning standard player
[error] MentalMap.cpp:96 getMyself(): Myself not found in mental map
[error] MentalMap.cpp:97 getMyself(): Returning standard player
[warning] GameScreen.cpp:82 drawScreen(): No camera frame received yet
[error] GameScreen.cpp:185 update(): Invalid image
```

I'd like to run more tests on Linux, too, to discard whether recently introduced changes may be the reason behind this. Targeted configurations:

* (webcam, no robot) `robotDevastation --mockRobotManager --yarpLocalImageManager`
* (robot with camera) `robotDevastation --robotName rd2`

The latter actually did work on Windows some time ago and was never tested again since then.

David-Estevez, 2017-04-10:
> yarp: Could not find carrier "mjpeg"

Did you compile the "mjpeg" carrier in YARP v2.3.68?

PeterBowman, 2017-04-10:
Probably not, and the user manual just states:
> cmake .. -DCREATE_OPTIONAL_CARRIERS=ON -DENABLE_yarpcar_mjpeg_carrier=ON  # Only required for mjpeg that should improve video comms

Anyway, I can't recall how I managed to make it work a couple of months ago, and this was already an issue with the old C opencv_grabber interface (still in master).

David-Estevez, 2017-04-10:
Just to discard that something might have gone wrong during your installation, could you try to run `yarp plugin mjpeg` ?

In my PC, the output was something like this:

```
$ yarp plugin mjpeg
Yes, this is a YARP plugin
  * library:        /usr/local/lib/yarp/yarp_mjpeg.so
  * system version: 5
  * class name:     yarp::os::MjpegCarrier
  * base class:     yarp::os::Carrier
```

jgvictores, 2017-04-10:
`mjpeg` is not required. It is the default carrier, but as can be seen on [these lines](https://github.com/asrob-uc3m/robotDevastation/blob/921f49d7c0585eac9e1478ef0cc4826524923d9c/src/libraries/ImageLib/YarpImageManager.cpp#L22-L61), we fall back to non-MJPEG (just YARP default) if the `mjpeg`connection cannot be made.
@David-Estevez What happened on your system is that it was detected as available, but a multiple-version installation issue resulted in segfaults.

jgvictores, 2017-04-10:
PD: For the local yarp image manager, it's actually [these lines](https://github.com/asrob-uc3m/robotDevastation/blob/921f49d7c0585eac9e1478ef0cc4826524923d9c/src/libraries/ImageLib/YarpLocalImageManager.cpp#L30-L38).

David-Estevez, 2017-04-10:
> @David-Estevez What happened on your system is that it was detected as available, but a multiple-version installation issue resulted in segfaults.

I see, could you document this in the developer manual, please? :smile:

PeterBowman, 2017-04-10:
> Just to discard that something might have gone wrong during your installation, could you try to run yarp plugin mjpeg ?

There is no `yarp_mjpeg.dll` anywhere. I'll perform more tests in a day or two, right now I'm not working on my Windows partition (and it's utterly slow to load).

PeterBowman, 2017-04-13:
> There is no yarp_mjpeg.dll anywhere.

And it couldn't be, YARP 2.3.68 doesn't even compile with `ENABLE_yarpcar_mjpeg` set to `ON` due to a missing include at `ACESemaphoreImpl.h` and [this](https://www.codeproject.com/Questions/369516/C-problem-in-basetsd-h).

PeterBowman, 2017-05-01:
> Anyway, I can't recall how I managed to make it work a couple of months ago, and this was already an issue with the old C opencv_grabber interface (still in master).

Per #90:

> the opencv_grabber plugin does not seem to work on this configuration (no problem on Linux); it's still using the old OpenCV 1 interface, we found out that the cv::Mat way of doing this solves the problem

PeterBowman, 2017-10-21:
This is the outcome of `robotDevastation --mockRobotManager --yarpLocalImageManager` at current `develop` (6de91db), Windows 10 64 bits, compiled on VS2017 (x32) with YARP 2.3.70.1, no `mjpeg` carrier:

```
--------------------------------------------------------------
Robot Devastation @ ASROB 2014 (C) Robotics Society of the Universidad Carlos III de Madrid
Welcome to Robot Devastation v0.2, developed by David Estevez and Juan G Victores.
Based on Robot Devastation v0.1, developed by Santiago Morante and Juan G Victores.
 ____       _           _   ____                      _        _   _             
|  _ \ ___ | |__   ___ | |_|  _ \  _____   ____ _ ___| |_ __ _| |_(_) ___  _ __  
| |_) / _ \| '_ \ / _ \| __| | | |/ _ \ \ / / _` / __| __/ _` | __| |/ _ \| '_ \ 
|  _ < (_) | |_) | (_) | |_| |_| |  __/\ V / (_| \__ \ || (_| | |_| | (_) | | | |
|_| \_\___/|_.__/ \___/ \__|____/ \___| \_/ \__,_|___/\__\__,_|\__|_|\___/|_| |_|

Fire with 'space'. Reload with 'r'. Move with 'Left, Up, Down, Right'. Run "robotDevastation --help" for help.
For a full description, please visit http://asrob.uc3m.es/rddoc/group__robotDevastation.html.
--------------------------------------------------------------
Insert player id [0]?>Insert name [player0]?>Insert team [0]?>Insert robotName [rd1]?>[error] AudioManager.cpp:113 rd::AudioManager::Register(): AudioManager with id "SDL" was already registered

Player information:
	-id: 0
	-name: player0
	-team: 0
	-robotName: rd1
[debug] StateDirector.cpp:31 rd::StateDirector::addTransition(): Added transition from InitState to GameState linked to condition 1
[debug] StateDirector.cpp:31 rd::StateDirector::addTransition(): Added transition from InitState to  linked to condition 2
[debug] StateDirector.cpp:31 rd::StateDirector::addTransition(): Added transition from GameState to DeadState linked to condition 1
[debug] StateDirector.cpp:31 rd::StateDirector::addTransition(): Added transition from GameState to  linked to condition 2
[debug] StateDirector.cpp:31 rd::StateDirector::addTransition(): Added transition from DeadState to GameState linked to condition 1
[debug] StateDirector.cpp:31 rd::StateDirector::addTransition(): Added transition from DeadState to  linked to condition 2
[debug] YarpStateDirector.cpp:19 rd::YarpStateDirector::Start(): Starting StateDirector for id InitState
yarp: Port /robotDevastation/0/rdServer/rpc:c active at tcp://10.11.12.1:10004/
yarp: Port /robotDevastation/0/rdServer/info:i active at tcp://10.11.12.1:10005/
yarp: Sending output from /robotDevastation/0/rdServer/rpc:c to /rdServer/rpc:s using tcp
[success] YarpNetworkManager.cpp:91 rd::YarpNetworkManager::start(): Connected robotDevastation RpcClient to rdServer RpcServer!
yarp: Receiving input from /rdServer/info:o to /robotDevastation/0/rdServer/info:i using tcp
[success] YarpNetworkManager.cpp:99 rd::YarpNetworkManager::start(): Connected from rdServer info to robotDevastation callbackPort!
[error] YarpNetworkManager.cpp:266 rd::YarpNetworkManager::keepAlive(): NetworkManager has not been started
===robotDevastation===
Number of players: 1
----------------------
ID: 0
Name: player0
Health: 100/100
Team ID: 0
Score: 0
[info] YarpNetworkManager.cpp:270 rd::YarpNetworkManager::keepAlive(): Keep alive...
===robotDevastation===
Number of players: 1
----------------------
ID: 0
Name: player0
Health: 100/100
Team ID: 0
Score: 0
[info] YarpNetworkManager.cpp:270 rd::YarpNetworkManager::keepAlive(): Keep alive...
===robotDevastation===
Number of players: 1
----------------------
ID: 0
Name: player0
Health: 100/100
Team ID: 0
Score: 0
[debug] InitState.cpp:120 rd::InitState::onKeyUp(): Key was pressed!
[debug] MockRobotManager.cpp:288 rd::MockRobotManager::setEnabled(): MockRobotManager disabled
[debug] YarpLocalImageManager.cpp:23 rd::YarpLocalImageManager::start(): Camera id: -1
[info] YarpNetworkManager.cpp:270 rd::YarpNetworkManager::keepAlive(): Keep alive...
===robotDevastation===
Number of players: 1
----------------------
[... 40 lines cut ...]

It gets stuck on the last line forever. Notice the `Connected to robot camera.` statement.

Sometimes, the server logs the user in and then I see the following output:

```
[success] YarpLocalImageManager.cpp:46 rd::YarpLocalImageManager::start(): Connected to robot camera.
[info] YarpNetworkManager.cpp:225 rd::YarpNetworkManager::login(): rdServer response from login: [ok]
[info] YarpNetworkManager.cpp:270 rd::YarpNetworkManager::keepAlive(): Keep alive...
===robotDevastation===
Number of players: 1
----------------------
ID: 0
Name: player0
Health: 100/100
Team ID: 0
Score: 0
[debug] YarpStateDirector.cpp:33 rd::YarpStateDirector::Stop(): Stopping StateDirector for id InitState
[debug] InitState.cpp:77 rd::InitState::cleanup(): Cleanup!!
[debug] YarpStateDirector.cpp:19 rd::YarpStateDirector::Start(): Starting StateDirector for id GameState
[debug] MockRobotManager.cpp:285 rd::MockRobotManager::setEnabled(): MockRobotManager enabled
[warning] GameScreen.cpp:91 rd::GameScreen::drawScreen(): No camera frame received yet
[error] GameScreen.cpp:194 rd::GameScreen::update(): Invalid image
...
```

PeterBowman, 2017-10-21:
Things to take a closer look upon and fix while using a local camera (`--yarpLocalImageManager`):

* check why server response is so erratic: should it allow login or reject us?
* if we are rejected, disable the `YarpLocalImageManager::onRead` callback and let the game close itself
* if we are spawned correctly, fix the `Invalid image` errors (assuming that plain YARP carriers should just work)

PeterBowman, 2017-10-23:
It looks like local camera support on Windows was broken at 7b60f0c: by changing the default `camera_id` to `0` (instead of `-1`), `opencv_grabber` is now able to configure the webcam. Note that this is how the device itself works if no `camera` option is passed, using the `CV_CAP_ANY` enumerator value (which resolves to `0`), see [docs](http://www.yarp.it/OpenCVGrabber_8cpp_source.html#l00059).

PeterBowman, 2017-10-23:
Concluding:

* No idea why the server was rejecting me sometimes; this is not a recurring problem and a new issue should be filed in case it reappears.
* Local image grabber works nicely on Windows with `camera_id` set to `0`. I'll check whether this change affects Linux installations and merge the patch otherwise.
* Remote image grabber was not tested, but should work. Anyway, it had no connection to the root cause of this issue.

Notes for the next RD meeting addressed to newcomers that want to compile the game on Windows:

* YARP 2.3.70.1 should work. I recall telling someone that newer releases may spoil the compilation process, but this should not affect us (until 2.3.72).
* In case you want to test the local camera, compilation from sources is needed (at least, I don't see the `yarp_opencv.dll` plugin at my installation folder when using pre-compiled binaries).
* *Official* binaries pulled from the YARP downloads page should be good enough for receiving images from the real robot. Thus, we are not forced to follow the full and painful compilation process. Not tested, though.
* To improve communications with the real robot, the `mjpeg` might be needed. I don't see its flag enabled for the pre-compiled binaries ([ref](https://github.com/robotology/yarp/blob/v2.3.70.1/scripts/admin/initial-cache.cmake)), thus refer to the manual for the usual YARP+dependencies build step.

commit 46d059bd (2017-10-23): Set default camera_id to '0' (as CV_CAP_ANY)

Closes #101.
closed 2017-10-23 by PeterBowman