asrob-uc3m/robotDevastation #106: Problema con Virtualbox

diegodevigo, 2017-09-22:
Estoy intentando instalar ubuntu en virtual box pero solo me deja la opcion de 32 bits, hay alguna forma de cambiarlo?

jgvictores, 2017-09-22:
Para instalar una máquina virtual de 64 bits, tiene que ser sobre un sistema operativo de 64 bits (supongo que esto lo tienes, se ve en Panel de Control -> Sistema/Detalles) y la versión de Virtual Box de 64 bits (revisa esto!!).

PeterBowman, 2017-10-05:
See http://www.fixedbyvonnie.com/2014/11/virtualbox-showing-32-bit-guest-versions-64-bit-host-os/#.WdZNK3Bx3CI

PeterBowman, 2017-11-04:
Conseguimos solucionar el problema en uno de los PCs habilitando la virtualización en la BIOS. TL;DR del artículo enlazado en el comentario anterior:

> If VirtualBox is only showing 32 bit versions in the Version list make sure:
>
> * Your Host OS is 64-bits
> * Intel Virtualization Technology and VT-d are both enabled in the BIOS
> * The Hyper-V platform is disabled in your Windows Feature list.

closed 2017-11-04 by PeterBowman