asrob-uc3m/robotDevastation #114: Doxygen updater script is broken

PeterBowman, 2017-12-01:
There is no longer a `Doxyfile` in `doc/`. Instead, we must generate it with CMake from `cmake/templates/Doxyfile.in`. Upon a successful configuration step, a `Doxyfile` is placed in the build directory, then `doxygen` will generate the HTML documentation in, say, `build/html/`. Our cron script obviously fails, see [update-dox.sh](https://github.com/asrob-uc3m/robotDevastation/blob/dbc19de5f1d72b71b16212242e68672cc9d38736/scripts/admin/update-dox.sh) (apart from that, we had to install CMake from packages, current available version is 3.2).

We don't want to be constrained by the CMake version installed on the server (nor bother to change it whenever the project requirements change). Proposal: either run CMake scripts from within the cron shell script (`cmake -p script.cmake`), thus it won't hit any `CMAKE_MINIMUM_REQUIRED()` command, or drop the CMake way altogether and restore previous functionality (will probably affect doc generation via CI builds).

PeterBowman, 2017-12-02:
I chose to drop the CMake configuration step and restore previous usage (we'll still keep a `make dox` target for packaging). The server is up again: http://asrob.uc3m.es/rddoc/index.html.

closed 2017-12-02 by PeterBowman

jgvictores, 2017-12-02:
Yes, this is a much easier solution. It's not so easy to run our CMake commands on the server, not worth the benefits given our current use.

commit 6c50e245 (2017-12-02): Restore old Doxyfile location, generate docs via target at root/doc

Mostly reverts 1b7c8c7, but now we are setting the working directory in
add_custom_target(). Fixes #114.