asrob-uc3m/robotDevastation #123: Drop RD macros in favor of color-debug ones

PeterBowman, 2018-02-12:
Now: [src/libraries/UtilsLib/Macros.hpp](https://github.com/asrob-uc3m/robotDevastation/blob/67d01e5481117b440e667fb7ec9ef3033329e707/src/libraries/UtilsLib/Macros.hpp) (e.g. `RD_INFO`, `RD_DEBUG`, etc.).

Proposal: [roboticslab-uc3m/color-debug](https://github.com/roboticslab-uc3m/color-debug) (e.g. `CD_INFO`, `CD_DEBUG`, etc.).

These are mostly the same CLI-output macros, although the latter provides convenient CMake options, too. However, upstream macros do not flush their output (5cf9629) - we noticed that this matters on Windows (at least, it does on AppVeyor CI builds).

jgvictores, 2018-02-12:
Duplicate of #57 ?

PeterBowman, 2018-02-12:
Yeah, sorry. I'll keep this one as it provides a better description.