asrob-uc3m/robotDevastation #126: scripts/gnome/robotDevastation.desktop has hard-coded paths

jgvictores, 2018-03-18:
(...).desktop has hard-coded paths

Learn from https://github.com/roboticslab-uc3m/asibot-configuration-files

Refs:

* [cmake/templates/asibot-apps.desktop.in](https://github.com/roboticslab-uc3m/asibot-configuration-files/blob/935efafd5589daa540e22a47f33916d10ff4f2a6/cmake/templates/asibot-apps.desktop.in)
* [CMakeLists.txt](https://github.com/roboticslab-uc3m/asibot-configuration-files/blob/935efafd5589daa540e22a47f33916d10ff4f2a6/CMakeLists.txt#L42-L51) (no install step yet)

PeterBowman, 2018-04-13:
See https://github.com/roboticslab-uc3m/questions-and-answers/issues/52.

jgvictores, 2018-04-13:
https://developer.gnome.org/integration-guide/stable/desktop-files.html.en