asrob-uc3m/robotDevastation #140: YarpNetworkManager logout(): Logout failed

imontesino, 2018-10-19:
```
[debug] GameState.cpp:175 onKeyDown(): Exit was triggered!
[debug] YarpStateDirector.cpp:38 Stop(): Stopping StateDirector for id GameState
yarp: Removing input from /rd2/img:o to /robotDevastation/rd2/img:i
[info] RobotDevastation.cpp:331 interruptModule(): [info] YarpNetworkManager.cpp:280 keepAlive(): Keep alive...
Closing program...
yarp: Removing output from /robotDevastation/0/rdServer/rpc:c to /rdServer/rpc:s
yarp: Removing input from /rdServer/info:o to /robotDevastation/0/rdServer/info:i
[info] YarpNetworkManager.cpp:254 logout(): Logout...
[info] YarpNetworkManager.cpp:259 logout(): rdServer response from logout: 
[error] YarpNetworkManager.cpp:264 logout(): Logout failed
[debug] YarpStateDirector.cpp:20 Start(): Null state. Exiting...
yarp: Removing output from /robotClient/rd2/rpc:c to /rd2/rpc:s
[info] SDLScreenManager.cpp:172 cleanupSDL(): Freeing resources...
[INFO]RFModule closing.
[INFO]RFModule finished.
```

PeterBowman, 2018-11-07:
Note that this failure was observed on a connection between the RD client and a real robot with outdated software. That is, this problem could vanish after upgrading the RasPi aboard RD2, which is close to being done. Cannot reproduce this on a local (fake) robot.

PeterBowman, 2018-11-29:
> this problem could vanish after upgrading the RasPi aboard RD2, which is currently in progress

Done at https://github.com/asrob-uc3m/rd2/issues/1#issuecomment-441367317. I'm unable to reproduce this on the real RD2 robot, thus closing as invalid. Please reopen if needed.

closed 2022-06-04 by PeterBowman